Xilinx/xup_vitis_network_example #94: Ping not working in "basic" design

pouya-haghi, 2022-11-16:
Hi @mariodruiz,

This is following from issue #93. Please let me just reiterate the problem. I'm using a `basic` design on Alveo U280 with one node. In my setup, each node has three interfaces connected to a switch (one 100G NIC, one 40G NIC, and an FPGA). I followed `vnx-basic.ipynb` but pinging to the FPGA doesn't work. Here are the inline answers to your questions:

> 1. Is link detected in the FPGA?

Yes, but running `print("Link interface 0 {}; link interface 1 {}".format(ol.cmac_0.link_status(),ol.cmac_1.link_status()))` throws an attribute error saying that it could not find cmac_1. This is because I built the design with `interface=0` (single interface). So, when I run `print("Link interface 0 {}".format(ol.cmac_0.link_status()))` it is fine and I will get a `True`.

> 2. Did you try running the arp discovery in the FPGA and checking the ARP table?

I ran arp discovery in the FPGA and then checked the ARP table, However, I found that this table is empty. I did it both before and after `ping` but I it was empty for both cases.

> 3. Did you try using arpping?

Yes I ran:
```
!arping $alveo_ipaddr
arping: Suitable device could not be determined. Please, use option -I.
```
So, I ran arping from 100G NIC (enp175s0np0)
```
!arping -I enp175s0np0 $alveo_ipaddr
ARPING 10.10.1.12 from 10.10.1.2 enp175s0np0
^CSent 754 probes (754 broadcast(s))
Received 0 response(s)
KeyboardInterrupt
```
However, it was waiting indefinitely and I had to press ctrl+c. I have also attached a file that contains network layer register map and other details. I would appreciate it if you could help me with this. Thank you vey much for your help and support!
[debug_alveo.txt](https://github.com/Xilinx/xup_vitis_network_example/files/10025914/debug_alveo.txt)

mariodruiz, 2022-11-17:
Hi @pouya-haghi,

If the ARP table is empty after ARP discovery and arpping from the host, this indicates that there is no communication/path between the two devices.

Are both devices connected point to point? Or via switch? I would suggest you try direct connection first.

pouya-haghi, 2022-11-17:
Thank you Mario for your response!
They are connected through a switch. Since my setup is CloudLab I cannot try a direct connection as I don't have access. But, sure, I will try to find a solution.

mariodruiz, 2022-11-17:
As I mentioned above, if either the ARP table is empty or ping does not work, these devices are not able to reach each other. This could be because they are in different VLANs in the switch or something else in the configuration.

pouya-haghi, 2022-11-17:
Yup, got it. Thank you again Mario for all of your help! Appreciate it!

closed 2022-11-17 by pouya-haghi

mariodruiz, 2022-11-18:
@pouya-haghi, when you find the solution please post it here, not the details but the high level. This can be useful for someone else facing the same problem

pouya-haghi, 2022-11-18:
Sure, of course.

pouya-haghi, 2023-01-02:
Hi @mariodruiz 

I just wanted to update you that the issue was that 100G NIC driver was not installed. After the installation, it worked properly. Thank you again for your help. 

Best,
Pouya