Xilinx/xup_vitis_network_example #98: CMAC long bring-up time after xbutil reset

TheFlux7, 2023-01-17:
Hello,

Thank you for this useful repository.

We are using a modified version of your vnx-benchmark-throughput-switch example.
We have 2 Alveo U55C, connected with fibers via a switch, with RS-FEC enabled.

We lose a lot of packets during the first test run after one card has been freshly reset. However, if we wait enough time (>5s) after enabling RS-FEC in CMAC, but before performing ARP Discovery, all packets are sent and received correctly.

Here is the execution sequence in more details:
1. HOT reset of one card
2. Download bitstreams
3. Enable CMAC RS-FEC
4. CMAC link status is instantaneously available
5. Sleep 10s
6. Set network layers' IP addresses
7. Set network layers' sockets
8. Populate network layers' socket table
9. Execute network layers' ARP discovery
10. Configure and start application

Without the sleep on step 5, packets are not transmitted successfully. With the sleep, everything works as expected.
We also tried to reset the CMAC after enabling RS-FEC, but then we need to wait even longer.

As sleeps are not reliable, we would like to understand the reasons why CMAC takes such a long time.
Do you happen to know what could be the cause for our observations?

Thanks and best regards
Bastien

mariodruiz, 2023-01-20:
Hi @TheFlux7,

> Do you happen to know what could be the cause for our observations?

I don't. In the current design, I can only recommend you try to check the register map of the CMAC to see if something is taking longer to get configured. You can try to ask this question on the Xilinx forums.

An alternative you can try, is to enable RS-FEC just after the xclbin is programmed. 
You can do this, by modifying this file to enable RS-FEC https://github.com/Xilinx/xup_vitis_network_example/blob/master/Ethernet/src/cmac_0_axi4_lite_user_if.v
Then, you need to rebuild the cmac.xo and the xclbin.

Mario

TheFlux7, 2023-01-23:
Hi @mariodruiz 

Thank you for your reply!

We conducted a small experiment:
We connected both cards directly, without the switch. It turns out that no packets are lost this way.
This means our switch needs some time to be ready. We'll investigate on our own for now.

Thank you for your time.

closed 2023-01-23 by TheFlux7