Xilinx/xup_vitis_network_example #101: Reducing CMAC/NetworkLayer Frequency

ChunshuWu, 2023-02-25:
Hi Mario, 

I've been trying to compile a design on Alveo U280 based on "benchmark". As the design grew bigger, timing became an issue. I reduced the frequency of user logic following issue #42 to 100 MHz, but here's another error: 

ERROR: [VPL 101-2] design did not meet timing - Design did not meet timing. One or more unscalable system clocks did not meet their required target frequency. For all system clocks, this design is using 0 nanoseconds as the threshold worst negative slack (WNS) value. List of system clocks with timing failure:
system clock: txoutclk_out[0]; slack: -0.031 ns

It seems the clock _txoutclk_out_ is related to CMAC/NetworkLayer, which is 322 MHz. Is there a way to reduce the frequency? 
I also found there are reference clocks with frequency 161.1328125 MHz for xcu50/xcu55c parts in Ethernet/bd_cmac.tcl, are they somehow related? I'm aware that I'm using xcu280 part though. 

Thanks!

mariodruiz, 2023-02-27:
Hi @ChunshuWu,

> I also found there are reference clocks with frequency 161.1328125 MHz for xcu50/xcu55c parts in Ethernet/bd_cmac.tcl, are they somehow related?

These are GT reference clocks.

> It seems the clock txoutclk_out is related to CMAC/NetworkLayer, which is 322 MHz. Is there a way to reduce the frequency?

No, this frequency is generated by the CMAC and it's the clock domain used for its operation, but the `cmac` kernel does a CDC shortly after the CMAC. 

You can try to tell `v++` to use a different Vivado implementation strategy. https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Using-the-vivado-and-advanced-Options

Or you can uncomment the line below, this will generate the xclbin even if the WNS is 2ns, you can make this number tighter.  

https://github.com/Xilinx/xup_vitis_network_example/blob/d6e9d338ccdcfed41ed72afdf8ca819c57e16fc2/Makefile#L162

A design with a WNS of -0.031 ns should work most of the times.

ChunshuWu, 2023-02-27:
Uncommenting the line in Makefile works well, haven't tried a different implementation strategy yet but I think this should suffice, thanks!

closed 2023-02-27 by ChunshuWu