Xilinx/xup_vitis_network_example #103: How to send packets to multiple destinations in basic design

pouya-haghi, 2023-04-18:
Hi @mariodruiz ,

This is more like a question rather than raising an issue about the design but your help and directions would be really appreciated. 

I have a question about `basic` design. Suppose that I have three nodes (one FPGA attached to a host and two 100G NICs attached to two other hosts). I want to send a packet from FPGA to NIC1 and then from FPGA to NIC2. I think I should populate the socket table with two entries (one entry for each NIC). But the problem is that how can I tell the FPGA to send the packet to NIC1 (instead of NIC2) and then later to send it to NIC2 (instead of NIC1)? I couldnt find such a flexibility in host .ipynb code (in other words, I can just call `mm2s.start` but I dont have control over which device I want to send to). I would appreciate it if you could help me on how can I do this.

Also, I have another question. Now, lets assume that this time we want the FPGA receives two packets (one from NIC1 and one from NIC2). Is it possible to know which device was sending the data when the packet is reached to the FPGA (can I distinguish who was the sender on the FPGA side)? I know that the user logic doesnt have this information but is there any workaround? Any direction is appreciated.

Thank you for your help in advance! 
Pouya

pouya-haghi, 2023-04-19:
I think I found the answer for my first question. If I'm not mistaken we can control it with the last argument of `mm2s` (`mm2s_wh = mm2s.start(mm2s_buf,size, 1)`). In this example, 1 means to use the second entry of socket table.

But still I dont the clue for the second question :((

mariodruiz, 2023-04-19:
Hi @pouya-haghi,

I am assuming that the Alveo is connected to a switch as well as the NICs, and this is not a point to point connection. 

The UDP layer has a 16 entry table, each entry corresponds to a socket, where you specify ports and IP. The switch should do the IP routing and send the packet to the appropriate destination. As you mention, the last parameter of the `mm2` kernel is the index of this table.

For the second question, the AXI4-Stream from the UDP layer to the user app contains the ID and metauser information. How this information is provided is described on the documentation https://github.com/Xilinx/xup_vitis_network_example/tree/master/NetLayers#udp

Mario

pouya-haghi, 2023-04-19:
Hi Mario,

Yes, that's correct. The alveo is connected to the switch as well and it's not a point to point connection.

Yup, got it. Thank you very much Mario!

closed 2023-04-19 by pouya-haghi