Xilinx/xup_vitis_network_example #104: Build problem for U50

danwetzel, 2023-04-24:
I am unable to build the xup_vitis_network_example, basic application for U50.  Similar to #83 but with a supported OS.

**Environment:**
```
lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	RedHatEnterprise
Description:	Red Hat Enterprise Linux release 8.6 (Ootpa)
Release:	8.6
Codename:	Ootpa
```
**Vitis Version:**
```
****** Xilinx Vitis Development Environment
****** Vitis v2022.2 (64-bit)
  **** SW Build 3671529 on 2022-10-13-17:52:08
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

```
**XRT Version:**
```
Version              : 2.14.384
Branch               : 2022.2
Hash                 : 090bb050d570d2b668477c3bd0f979dc3a34b9db
Hash Date            : 2022-12-09 08:55:04
XOCL                 : 2.14.384, 090bb050d570d2b668477c3bd0f979dc3a34b9db
XCLMGMT              : 2.14.384, 090bb050d570d2b668477c3bd0f979dc3a34b9db
```
**Command:**
```make all DEVICE=xilinx_u50_gen3x16_xdma_5_202210_1 INTERFACE=0 DESIGN=basic```

**Resulting Errors:**
```
ERROR: [VPL 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified
ERROR: [VPL 19-3477] Update of parameter 'PARAM_VALUE.CHANNEL_ENABLE' failed for IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt'. ERROR: [Common 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified

ERROR: [VPL 19-3428] Failed to create Customization object ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt
ERROR: [VPL 17-70] Application Exception: Failed to create subcore IP 'cmac_bd_cmac_uplus_0_0_gt'. Create IP failed with errors
ERROR: [VPL 19-167] Failed to deliver one or more file(s).
ERROR: [VPL 19-3541] IP Elaboration error: Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs: 
ERROR: [VPL 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified
ERROR: [VPL 19-3477] Update of parameter 'PARAM_VALUE.CHANNEL_ENABLE' failed for IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt'. ERROR: [Common 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified

ERROR: [VPL 19-3428] Failed to create Customization object ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0/cmac_bd_cmac_uplus_0_0_gt
ERROR: [VPL 17-70] Application Exception: Failed to create subcore IP 'cmac_bd_cmac_uplus_0_0_gt'. Create IP failed with errors
ERROR: [VPL 19-167] Failed to deliver one or more file(s).
ERROR: [VPL 19-3541] IP Elaboration error: Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs: 
ERROR: [VPL 19-98] Generation of the IP CORE failed.
Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs: 
Could not configure the upgraded 'IP cmac_bd_cmac_uplus_0_0'.
If the problem persists, please contact Xilinx support.
Failed to generate IP 'ulp_cmac_0_0/cmac_bd_cmac_uplus_0_0'. Failed to generate 'Elaborate Sub-Cores' outputs: 
ERROR: [VPL 41-1030] Generation failed for the IP Integrator block cmac_0 
ERROR: [VPL 60-773] In '/home/dw4203/gitrep/xup/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/runme.log', caught Tcl error:  ERROR: [Common 17-39] 'generate_target' failed due to earlier errors.
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, Step failed: generate_target An error stack with function names and arguments may be available in the 'vivado.log'.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
WARNING: [VPL 60-1142] Unable to read data from '/home/dw4203/gitrep/xup/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/output/generated_reports.log', generated reports will not be copied.
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [14:22:35] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:05 ; elapsed = 00:01:32 . Memory (MB): peak = 651.434 ; gain = 0.000 ; free physical = 14032 ; free virtual = 36318
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:106: basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/vnx_basic_if0.xclbin] Error 1
```

mariodruiz, 2023-04-25:
Hi @danwetzel,

What is the output of `locale`?

Mario

danwetzel, 2023-04-26:
The output is:
```LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

C1bergh0st, 2024-07-02:
I realize this issue has been up for a while but where you (@danwetzel) able to solve it?

C1bergh0st, 2024-07-02:
The Issue with Alveo U50 compilation (*probably*) arises from a wrong part number.
There are 2 types of Alveo U50 boards: Normal and Low-Voltage.
The associated fpga part numbers are (taken from the [datasheet](https://docs.amd.com/r/en-US/ug1371-u50-reconfig-accel/Creating-a-Vivado-RTL-Project)):
XCU50-FSVH2104-2-E (Normal)
XCU50-FSVH2104-2LV-E (Low Voltage)

There was an issue raised here about the Xilinx Board Store having a typo [here](https://github.com/Xilinx/XilinxBoardStore/issues/150)

i suspect this typo (xcu50-fsvh2104-2L-e instead of xcu50-fsvh2104-2-e) was thus repeated in the tcl files under /Ethernet/:
https://github.com/Xilinx/xup_vitis_network_example/tree/0b64f7c30c5006d6759262a5de4fca7ed5087d12/Ethernet

and in turn caused these problems.
I replaced all occurrences of xcu50-fsvh2104-2L-e with xcu50-fsvh2104-2-e and this fixed this error:
`ERROR: [VPL 17-69] Command failed: Invalid site type qualifier 'GTHE4_CHANNEL' specified`
The compilation is currently still running, i will update once i have a finished output and again when i am able to load it on an alveo u50.

For completeness:
### OS version
```
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
```
### Vitis version
```
****** Xilinx Vitis Development Environment
****** Vitis v2022.2 (64-bit)
  **** SW Build 3671529 on 2022-10-13-17:52:08
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

```
### XRT version
```
Version              : 2.16.204
Branch               : 2023.2
Hash                 : fa4c0045003fed0acea4593788dce5ef6d0b66ee
Hash Date            : 2023-10-11 23:45:57
XOCL                 : unknown, unknown
XCLMGMT              : unknown, unknown
```

mariodruiz, 2024-07-02:
If the change works, can you please open a PR?

C1bergh0st, 2024-07-02:
Created #129 
i think i screwed up some of the file endings though...
Change works, compiles and produces loadable .xclbin
Still need to do some functionality tests if everything works, that will take a while since our setup needs a few hardware changes.

closed 2024-08-01 by mariodruiz