Xilinx/xup_vitis_network_example #108: Bug reported: name `ol` is not defined.

zhangchenqi123, 2023-08-16:
In `vnx-basic-image-transfer.ipynb` 
In section `Configure the application`
The `ol` is not defined.

[image]

[image]

This is the first time `ol`  has appeared here, and I haven't found where it is defined before.

mariodruiz, 2023-08-16:
Hi @zhangchenqi123 

Thank you for reporting this.
It should be either `ol_w1` or `ol_w0` and then in the assignation they the object should be used appropriately. 

Would you be able to do a PR with the changes?

zhangchenqi123, 2023-08-16:
Okay, I'd be happy to do this, but I'm fixing another bug. After I finish fixing it, I will do a PR.

zhangchenqi123, 2023-08-17:
Aside from this issue, I've also come across another problem that I believe warrants opening a separate issue. I hope this doesn't inconvenience you in any way.

mariodruiz, 2023-08-17:
can you please share here the changes you have attempted? 

Does any of the notebooks work?

mariodruiz, 2023-08-17:
I think that portion should be replace with 

```python
mm2s_remote = ol_w1.krnl_mm2s_1
s2mm_local = ol_w0.krnl_s2mm_1

size = img_height * img_width * 3

if hasattr(ol_w1, 'HBM0'):
    s2mm_bank = ol_w0.HBM0
    mm2s_bank = ol_w1.HBM0
elif hasattr(ol, 'bank2'):
    s2mm_bank = ol_w0.bank1
    mm2s_bank = ol_w1.bank2
else:
    s2mm_bank = ol_w0.bank1
    mm2s_bank = ol_w1.bank1
```

zhangchenqi123, 2023-08-18:
Ok, that works. Thanks a lot.
Besides, I found another bug in the last but one cell of this notebook.
[image]
It should be `tx`, because on the tx end, the bytes are 'out', on the rx end, the bytes are 'in'.
I will do a pr to modify this, too.

closed 2023-08-18 by zhangchenqi123