Xilinx/xup_vitis_network_example #114: Nolink when porting to ADM-PCIE-9V3

liubenyuan, 2023-09-16:
Hi I tried to build a simple lookback example on ADM-PCIE-9V3, where only CMAC and network-stack-core are kept, the PCIE part are removed. The block diagram is as follows:

### Build Issues

```bash
$ vitis --version

****** Xilinx Vitis Development Environment
****** Vitis v2022.2 (64-bit)
  **** SW Build 3671529 on 2022-10-13-17:52:08
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
```

```bash
$ uname -a
Linux archlinux 6.5.3-native_amd-xanmod1-1 #1 SMP PREEMPT_DYNAMIC Thu, 14 Sep 2023 10:21:49 +0000 x86_64 GNU/Linux
```

mariodruiz, 2023-09-16:
Hi @liubenyuan,

This board is not supported in this project. I suggest you seek support on the Xilinx forums.

liubenyuan, 2023-09-16:
> Hi @liubenyuan,
> 
> This board is not supported in this project. I suggest you seek support on the Xilinx forums.

Thank you. One more question, is it ok to build a RTL only bitstream .bit like figure 1? Without configuration to the AXI4Lite registers of 100G-network-core and the CMAC?

closed 2023-09-16 by liubenyuan

mariodruiz, 2023-09-17:
> One more question, is it ok to build a RTL only bitstream .bit like figure 1? Without configuration to the AXI4Lite registers of 100G-network-core and the CMAC?

Technically yes, but it is recommended to have the AXI4-Lite interface for debug and configuration.