Xilinx/xup_vitis_network_example #120: Can a 10G or 25G NIC card be used to execute and implement this design with Alveo u50 card?

lizajoseph, 2024-02-01:
I am trying to implement XUP Vitis network example design targeting on Alveo u50 platform. 
But we only have the following 2 NIC cards. Will I be able to implement and validate this design with any one of these NIC cards?

1) HPE Ethernet 10Gb 2-port 562SFP+ Adapter
2) Solarflare X2522 25G 2-port 

Thanks,
Liza

mariodruiz, 2024-02-01:
Hi @lizajoseph,

No, you need the Alveo connected to a 100G network equipment.

lizajoseph, 2024-02-01:
Thanks for your response.
We do not have a 100G switch or a 100G NIC card. Is there any alternative way to connect with the Alveo u50 card?

mariodruiz, 2024-02-01:
You may be able to use breakout cables, but this project won't support this. I am not aware of any project that will enable you with this.

lizajoseph, 2024-02-01:
Okay, Thanks Mario for giving me this clarity.