Xilinx/xup_vitis_network_example #121: Compiler error with Alveo xilinx_u50_gen3x16_xdma_5_202210_1 with Vivado and Vitis Version 2023.1 and XRT xrt_202310.2.15.225_20.04-amd64-xrt

lizajoseph, 2024-02-14:
**For Vivado questions, please use [Vivado forum](
https://forums.xilinx.com/t5/Vivado-RTL-Development/ct-p/DESIGN)**
ERROR: [VPL 8-11365] for the instance 'network_layer_bd_i' of module 'network_layer_bd' declared at /xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/b30f/src/network_layer_bd.v:13' ERROR: [VPL 8-11365] for the instance 'network_layer_bd_i' of module 'network_layer_bd' declared at

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo)**

**For pynq questions, please use [the PYNQ discussion forum](
https://discuss.pynq.io/).**

**Usign Vitis 2021.2 or older? Make sure the [Y2K22 patch is applied](
https://support.xilinx.com/s/article/76960?language=en_US)**

If you still want to raise an issue here, please give us as much detail as
possible to the issue you are seeing. We have listed some helpful fields below.

- Please, use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.

### Build Issues

1. OS version, e.g. `lsb_release -a`
1. Vitis version `vitis -version`
   1. If Vitis is 2021.2 or older. Is the [Y2K22 patch applied](https://support.xilinx.com/s/article/76960?language=en_US)?
1. XRT version `xbutil version`

### Run Time Issues

1. OS version `lsb_release -a`
1. XRT version `xbutil version`
1. pynq version `pynq version`
1. JupyterLab and Dask version if applicable

lizajoseph, 2024-02-14:
Even after sourcing the Vitis and Vivado 2023.1 settings64.sh and XRT xrt_202310.2.15 getting below compile error for Alveo xilinx_u50_gen3x16_xdma_5_202210_1
ERROR: [VPL 8-11365] for the instance 'network_layer_bd_i' of module 'network_layer_bd' declared at '/home/liza.joseph@CORP.FLOWEDGE.IN/XUP_Vitis_Network_example/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/b30f/src/network_layer_bd.v:13', named port connection 'ap_clk' does not exist [/home/liza.joseph@CORP.FLOWEDGE.IN/XUP_Vitis_Network_example/xup_vitis_network_example/basic.intf0.xilinx_u50_gen3x16_xdma_5_202210_1/link/vivado/vpl/prj/prj.gen/my_rm/bd/ulp/ipshared/b30f/src/networklayer.v:99]

mariodruiz, 2024-02-14:
Hi @lizajoseph,

Please provide this information 

## Build Issues
1. OS version, e.g. lsb_release -a
1. Vitis version vitis -version
1. XRT version xbutil version

lizajoseph, 2024-02-14:
1) OS version - Ubuntu 20.04
2) Vitis version - 2023.1
3) XRT version - 2.15.225

I am trying to build the basic design for Alveo xilinx_u50_gen3x16_xdma_5_202210_1

Command the run the basic XUP vitis example
make all DEVICE=xilinx_u50_gen3x16_xdma_5_202210_1 INTERFACE=0 DESIGN=basic

mariodruiz, 2024-02-14:
I was able to generate the design from a fresh copy of the repo without problems.

Can you please try that?

```sh
make all DEVICE=xilinx_u50_gen3x16_xdma_5_202210_1.xpfm DESIGN=basic 2>&1 | tee build.log
```

If it does not work, please share the build.log

lizajoseph, 2024-02-15:
With a fresh repo copy and with the command you have shared, I am able to generate the .xclbin

make all DEVICE=xilinx_u50_gen3x16_xdma_5_202210_1.xpfm DESIGN=basic 2>&1 | tee build.log

mariodruiz, 2024-02-15:
Resolved

closed 2024-02-15 by mariodruiz