Xilinx/xup_vitis_network_example #122: Alveo u50 card to NIC card physical link is not getting detected

lizajoseph, 2024-02-19:
**For Vivado questions, please use [Vivado forum](
https://forums.xilinx.com/t5/Vivado-RTL-Development/ct-p/DESIGN)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo)**

**For pynq questions, please use [the PYNQ discussion forum](
https://discuss.pynq.io/).**

**Usign Vitis 2021.2 or older? Make sure the [Y2K22 patch is applied](
https://support.xilinx.com/s/article/76960?language=en_US)**

If you still want to raise an issue here, please give us as much detail as
possible to the issue you are seeing. We have listed some helpful fields below.

- Please, use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.

### Build Issues

1. OS version, e.g. `lsb_release -a`
1. Vitis version `vitis -version`
   1. If Vitis is 2021.2 or older. Is the [Y2K22 patch applied](https://support.xilinx.com/s/article/76960?language=en_US)?
1. XRT version `xbutil version`

### Run Time Issues

1. OS version `lsb_release -a`
1. XRT version `xbutil version`
1. pynq version `pynq version`
1. JupyterLab and Dask version if applicable

lizajoseph, 2024-02-19:
After loading the .xclbin, I check for the physical link status between my Alveo u50 and the NIC card but the 'cmac_link' returns false value. But the NIC card is getting detected individually. Could you let me know why this link is not getting detected.

Attached screenshot of the validation steps in Jupyter Lab
[image]

mariodruiz, 2024-02-19:
Hi @lizajoseph,

To check the link you should use `ethtool`

```sh
sudo ethtool <interface name>
```

This should report the stats on the interface, including link status. From the report you show, it does not look like the interface has link.

lizajoseph, 2024-02-20:
Will the link status be successful only if the network equipment is 100G NIC? or the link status should be able to detect any network equipment connected to the Alveo u50 card?

mariodruiz, 2024-02-20:
Only 100G network equipment would work.

lizajoseph, 2024-02-20:
Since auto negotiation is not configured as part of the design it will not be able to exchange information between two different devices (u50 card programmed with CMAC with 100G Ethernet IP and a 25G Solarflare NIC card that I have connected with the u50 card) which are not compatible. Is my understanding correct?

mariodruiz, 2024-02-20:
Yes. I mentioned this here https://github.com/Xilinx/xup_vitis_network_example/issues/120

lizajoseph, 2024-02-20:
Thanks, you can close the ticket

closed 2024-02-20 by mariodruiz