Xilinx/xup_vitis_network_example #123: Can the XUP vitis network design work with 10G/25G Ethernet IP instead of the existing 100GEthernet IP ?

lizajoseph, 2024-02-22:
**For Vivado questions, please use [Vivado forum](
https://forums.xilinx.com/t5/Vivado-RTL-Development/ct-p/DESIGN)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo)**

**For pynq questions, please use [the PYNQ discussion forum](
https://discuss.pynq.io/).**

**Usign Vitis 2021.2 or older? Make sure the [Y2K22 patch is applied](
https://support.xilinx.com/s/article/76960?language=en_US)**

If you still want to raise an issue here, please give us as much detail as
possible to the issue you are seeing. We have listed some helpful fields below.

- Please, use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.

### Build Issues

1. OS version, e.g. `lsb_release -a`
1. Vitis version `vitis -version`
   1. If Vitis is 2021.2 or older. Is the [Y2K22 patch applied](https://support.xilinx.com/s/article/76960?language=en_US)?
1. XRT version `xbutil version`

### Run Time Issues

1. OS version `lsb_release -a`
1. XRT version `xbutil version`
1. pynq version `pynq version`
1. JupyterLab and Dask version if applicable

lizajoseph, 2024-02-22:
Does this design support 10G Ethernet IP targeted on Alveo u50? 
- I observe that the output from the 100G Ethernet IP CMAC kernel is 512 bit AXI4 stream 
whereas for 10G Ethernet the output is 64 bit. 
- The CDC logic to convert from kernel clock to the 100G Ethernet subsystem clock can also support for 10G ethernet subsystem?
- The No. of lanes for 100G Ethernet is {4x25}, can this be modified to support for 10G ethernet?
- Is the 100G ethernet subsystem register mapping compatible for 10G Ethernet IP?
These are some of the differences I observe between the 100G and 10G Ethernet subsystem, are there more differences which I need to make note of for further implementation or changes?

mariodruiz, 2024-02-22:
Hi @lizajoseph,

This repository and associated IP will only work for 100 G. There is no current plan to support any other speed. 
There is a substantial body of work to support 10G.  I suggest you check the 10/25G Ethernet IP, the modifications to this project are not trivial. 

The CMAC kernel as it is designed here will not work for 10G.

lizajoseph, 2024-02-22:
Okay, Thanks for your confirmation.

closed 2024-04-27 by mariodruiz