Xilinx/xup_vitis_network_example #124: Compiling issue during routing phase with U250

GiacomoLevrini, 2024-03-01:
Hello, 

I am tryng to build the default version of the network, launching make in the main directory. 
The settings I am using are:
- DEVICE = xilinx_u250_gen3x16_xdma_4_1_202210_1
- INTERFACE = 0
- DESIGN = benchmark

During the build, I end up getting this error:
``` Routing results verification failed due to partially-conflicted nets (Up to first 10 of violated nets):  level0_i/level1/level1_i/ulp/ss_ucs/inst/aclk_kernel_00_hierarchy/clock_throttling_aclk_kernel_00/U0/Clk_Int_Cont level0_i/level1/level1_i/ulp/ss_ucs/inst/aclk_kernel_01_hierarchy/clock_throttling_aclk_kernel_01/U0/Clk_Int_Cont ``` 

The specifics of the machine I am using are:
 - OS version: CentOS Linux release 7.7.1908 
- Vitis version : Vitis v2022.2 (64-bit)
- XRT version :  2.14.354

How do I fix this?

mariodruiz, 2024-03-04:
Hi @GiacomoLevrini,

I was able to reproduce the issue. I am investigating and I will get back to you

mariodruiz, 2024-03-12:
@GiacomoLevrini is there any change you can use Vitis 2023.1?

GiacomoLevrini, 2024-03-12:
I think I could try to upgrade it. I'll let you know if something changes.

mariodruiz, 2024-03-12:
I tried with 2023.1 and the generation is successful.

commit 0b64f7c3 (2024-04-27): Update xilinx_u250_gen3x16_xdma_4_1_202210_1 build instructions

Fix #124
closed 2024-04-27 by mariodruiz