Xilinx/xup_vitis_network_example #130: RuntimeError: No Devices Found whn

SophiaLcy, 2024-10-24:
**For Vivado questions, please use [Vivado forum](
https://forums.xilinx.com/t5/Vivado-RTL-Development/ct-p/DESIGN)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v)**

**For Vitis questions, please use [the Vitis forum](
https://forums.xilinx.com/t5/Alveo-Accelerator-Cards/bd-p/alveo)**

**For pynq questions, please use [the PYNQ discussion forum](
https://discuss.pynq.io/).**

**Usign Vitis 2021.2 or older? Make sure the [Y2K22 patch is applied](
https://support.xilinx.com/s/article/76960?language=en_US)**

If you still want to raise an issue here, please give us as much detail as
possible to the issue you are seeing. We have listed some helpful fields below.

- Please, use [code snippets](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) to provide textual content instead of images.

### Build Issues

1. OS version, e.g. `lsb_release -a`
1. Vitis version `vitis -version`
   1. If Vitis is 2021.2 or older. Is the [Y2K22 patch applied](https://support.xilinx.com/s/article/76960?language=en_US)?
1. XRT version `xbutil version`

### Run Time Issues

1. OS version `lsb_release -a`
1. XRT version `xbutil version`
1. pynq version `pynq version`
1. JupyterLab and Dask version if applicable

SophiaLcy, 2024-10-24:
sorry I was uploading the image which was really slow T_T

mariodruiz, 2024-10-24:
Do you have PYNQ and XRT installed in both machines?

SophiaLcy, 2024-10-24:
> Do you have PYNQ and XRT installed in both machines?

yes when I type "which pynq"  it show
  `/rshome/*****/miniconda3/envs/xup-vitis/bin/pynq`

mariodruiz, 2024-10-24:
is this in both machines? Can you please comment about your hardware configuration?

SophiaLcy, 2024-10-25:
I installed XRT on both machines. However, I installed PYNQ on an NFS directory, so they share the same one, and I only had to install it once. The host's hardware configuration is attached in the original comment; the other is as follows.
[image]

mariodruiz, 2024-10-28:
It's been a while since I have not touched that part, but this could be useful: https://docs.dask.org/en/latest/deploying-ssh.html

SophiaLcy, 2024-10-29:
> What is the output when you print the `client`?
> 
> ```python
> from dask.distributed import Client, get_client
> 
> client = Client(<your connection>)
> client
> ```

this is the output

[image]

mariodruiz, 2024-10-29:
All the dask config looks good. Does dask have access to the same env as PYNQ?

Have you tried using PYNQ and checking if the device is visible? The issues seems like a env problem.

mariodruiz, 2025-05-08:
Hi @SophiaLcy any update?