Xilinx/xup_vitis_network_example #131: How to execute the basic kernel (image transfer) with C/C++

shibizhao, 2024-10-31:
Hi,

I am trying to execute the basic image transfer application with XRT in C/C++.

One U200 and one U280 cards are on my server. And I can transfer the image correctly using PYNQ with multi-thread rather than dask. So I would like to take the similar way in C/C++.

Currently, I can set the parameters for networklayer_0 (kernel) using `read_register` and `write_register` interfaces (in XRT native APIs). In order to validate the correctness: I reset the two cards first; and then execute the parameters settings for networklayer_0 using C/C++; and then execute the PYNQ program (only with the buffer allocation, kernel execution, and result validation.

Now, I would like to develop the kernel execution such as:
```
void recv_func2(xrt::kernel & krnl, xrt::bo & output_bo, size_t size){
  auto recv_run = krnl(output_bo, size);
  auto z = recv_run.wait();
  std::cout << z << std::endl;
}

void send_func2(xrt::kernel & krnl, xrt::bo & input_bo, size_t size, unsigned int id){
  auto send_run = krnl(input_bo, size, id);
}

int main(){
  xrt::device u200 = xrt::device(0);
  auto u200_name = u200.get_info<xrt::info::device::name>();
  std::cout << "[multi.cpp] device " << 0 << "'s name: " << u200_name << std::endl;

  xrt::device u280 = xrt::device(1);
  auto u280_name = u280.get_info<xrt::info::device::name>();
  std::cout << "[multi.cpp] device " << 1 << "'s name: " << u280_name << std::endl;

  auto u200_uuid = u200.load_xclbin("../u200/basic.intf3.xilinx_u200_gen3x16_xdma_2_202110_1/vnx_basic_if3.xclbin");
  auto u280_uuid = u280.load_xclbin("../u280/basic.intf3.xilinx_u280_gen3x16_xdma_1_202211_1/vnx_basic_if3.xclbin");

  auto krnl_u200 = xrt::kernel(u200, u200_uuid, "krnl_s2mm:{krnl_s2mm_0}");
  auto krnl_u280 = xrt::kernel(u280, u280_uuid, "krnl_mm2s:{krnl_mm2s_0}");

  auto input_bo = xrt::bo(u280, 1 << 27, krnl_u280.group_id(0));
  auto output_bo = xrt::bo(u200, 1 << 27, krnl_u200.group_id(0));

  size_t size = 1 << 27;

  std::vector<unsigned char> input_data(size);
  for(int i = 0; i < size; ++i){
    input_data[i] = i % 256 + rand() % 256;
  }

  std::vector<unsigned char> output_data(size);
  for(int i = 0; i < size; ++i){
    output_data[i] = 0;
  }

  input_bo.write(input_data.data());
  input_bo.sync(XCL_BO_SYNC_BO_TO_DEVICE);

  std::thread sendThread(send_func2, std::ref(krnl_u280), std::ref(input_bo), size, 1);
  std::thread recvThread(recv_func2, std::ref(krnl_u200), std::ref(output_bo), size);

  sendThread.join();
  recvThread.join();
}
```
This program passed the compilation. And there is no error during the execution (the return value of recv_run.wait() is 4). But the output_data is empty.

So I would like to ask whether I can achieve the data transfer via two FPGAs by this way.

By the way, I would also like to know that in your experiments, what is the bandwidth of this application? I measured the bandwidth using PYNQ through a super-large image (192MB) from U280 to U200. The bandwidth is only about 27Gb/s. I am confused about this result.

Thanks for your time!

mariodruiz, 2024-10-31:
Hi @shibizhao,

I think you should start the `recvThread` first. I assume you have configured the network and populated the socket table in advance.

> By the way, I would also like to know that in your experiments, what is the bandwidth of this application? I measured the bandwidth using PYNQ through a super-large image (192MB) from U280 to U200. The bandwidth is only about 27Gb/s. I am confused about this result.

How exactly are you measuring this? There's a benchmark design that should be used for performance metrics

shibizhao, 2024-11-01:
Hi, thanks for your reply.

I have changed the order of sendThread and recvThread, but the results are the same. There is no change in the output buffer object.

I use the "time.time()" to for the measurement:

mariodruiz, 2024-11-01:
I don't understand why you're providing Python code now.

- Where are you setting up the IP of the different Alveo cards?
- Where are you populating the socket table?
- Are you checking if the link is up?
- There are plenty of embedded proves, I suggest you query them to see where the packets are being dropped. https://github.com/Xilinx/xup_vitis_network_example/tree/master/NetLayers#embedded-probes
 

The basic design should not be used to measure performance. The transfer time via PCIe and the synchronization time will impact the bandwidth. You should use the benchmark design for this.

shibizhao, 2024-11-26:
Hi,

Sorry for the late reply and the confused python code I provided.
Here is my c++ code for the data transfer.

```
#include "experimental/xrt_bo.h"
#include "experimental/xrt_device.h"
#include "experimental/xrt_kernel.h"
#include "experimental/xrt_xclbin.h"
#include "experimental/xrt-next.h"
#include "experimental/xrt_ip.h"
#include <iostream>
#include <vector>
#include <map>
#include <cassert>
#include <cstdlib>
#include <string>
#include <fstream>
#include <iostream>
#include <iomanip>
#include <sstream>
#include <thread>
#include <chrono>
#include <pthread.h>
#include <thread>
#include <unistd.h>  
#include <stdio.h>
#include <sys/types.h> 
#include <sys/wait.h>

#define ALVEO_MAC_ADDR_OFFSET 0x0010
#define ALVEO_IP_ADDR_OFFSET 0x0018

#define ALVEO_UDP_NUMBER_SOCKETS_OFFSET 0x0810

#define ALVEO_UDP_THEIR_IP_ADDR_OFFSET 0x0820
#define ALVEO_UDP_THEIR_PORT_OFFSET 0x08a0
#define ALVEO_UDP_MY_PORT_OFFSET 0x0920
#define ALVEO_UDP_VALID_OFFSET 0x09a0

#define ALVEO_ARP_DISCOVERY 0x1010
#define ALVEO_ARP_VALID_OFFSET 0x1100
#define ALVEO_ARP_IP_ADDR_OFFSET 0x1400
#define ALVEO_ARP_MAC_ADDR_OFFSET 0x1800

constexpr std::size_t rsfec_config_ind_corr = 0x1000;
constexpr std::size_t rsfec_config_enable = 0x107C;
constexpr std::size_t reset_reg = 0x0004;

const int MAXIMUM_UDP_SOCKETS = 128;

struct socket{
  unsigned int their_ip;
  unsigned int their_port;
  unsigned int my_port;
  unsigned int valid;
  socket(){
    their_ip = 0;
    their_port = 0;
    my_port = 0;
    valid = 0;
  };
  socket(unsigned int their_ip, 
         unsigned int their_port, 
         unsigned int my_port, 
         unsigned int valid):
         their_ip(their_ip), their_port(their_port), 
         my_port(my_port), valid(valid){};
  socket(const socket& s){
[... 207 lines cut ...]
    output_data[i] = 0;
  }

  input_bo.write(input_data.data());
  input_bo.sync(XCL_BO_SYNC_BO_TO_DEVICE);

  std::thread recvThread(recv_func2, std::ref(krnl_u200), std::ref(output_bo), size);
  std::thread sendThread(send_func2, std::ref(krnl_u280), std::ref(input_bo), size, 0);
  
  recvThread.join();
  sendThread.join();
  

  output_bo.sync(XCL_BO_SYNC_BO_FROM_DEVICE);
  output_bo.read(output_data.data());

  for(int i = 0; i < 10; ++i){
    std::cout << "Data: " << (int)output_data[i] << " " << (int)input_data[i]<< std::endl;
  }

  return 0;
}
```

Currently, I think the recv function must wait after the end of the send function, but I find that it returned 4 very quickly.

shibizhao, 2024-11-28:
I have added the debug code to show the probes, and there is no any packet from the sender or to the receiver.